EvolutionGym/evogym #1: Suggestion for learning algorithms or library?

JunHill, 2022-01-31:
Hi, thank you for such interesting benchmark! 
I am doing some school project relating Deep Reinforcement Learning and Evolutionary algorithm. Could you suggest some more existing algorithms that would be interesting to try on this benchmark?

jagdeepsb, 2022-02-01:
Hi @JunHill, thanks for your question!

We have outlined several future research directions in the conclusion of our [paper](https://arxiv.org/pdf/2201.09863.pdf) (page 10).

Some broad strategies include: concurrently co-optimizing the design and control (in our work we use a bi-level optimization scheme), neuroevolution algorithms, morphogenetic development, gradient-based methods for design optimization, algorithms with decentralized controllers, and multi-task or multi-objective robot co-design algorithms.

JunHill, 2022-02-01:
Thanks for the keywords. I'll close the issue!

closed 2022-02-01 by JunHill