EvolutionGym/evogym #2: error C2248 when run python setup.py install

JunHill, 2022-02-01:
Hi, I'm facing an issue with building the simulator in Windows 10. 
After cloning the git repo, I installed Python dependencies with conda as instructed:
> conda env create -f environment.yml

Then I ran the following command to build the simulator:
> python setup.py install
and end up with this error:
> subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '/m']' returned non-zero exit status 1.

I scrolled up a bit and see the following lines in red:

```
  PythonBindings.cpp
  Robot.cpp
C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.cpp(271,3): error C2248: 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref': cannot access private member declared
 in class 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\externals\eigen\Eigen\src/Core/Ref.h(299): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.h(95): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.cpp(282,3): error C2248: 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref': cannot access private member declared
 in class 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\externals\eigen\Eigen\src/Core/Ref.h(299): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.h(95): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.cpp(294,3): error C2248: 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref': cannot access private member declared
 in class 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\externals\eigen\Eigen\src/Core/Ref.h(299): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.h(95): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]
  Sim.cpp

C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.cpp(309,3): error C2248: 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref': cannot access private member declared
 in class 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\externals\eigen\Eigen\src/Core/Ref.h(299): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>::Ref' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]

C:\MyPath\evogym\evogym\simulator\SimulatorCPP\Environment.h(95): message : see declaration of 'Eigen::Ref<Eigen::MatrixXd,0,Eigen::OuterStride<-1>>' [C:\MyPath\evogym\build\temp.win-amd64-3.7\Release\SimulatorCPP\simulator_cpp.vcxproj]
```

yunshengtian, 2022-02-01:
Have you run `conda activate evogym` before running `python setup.py install`?
Also, do you have cmake installed beforehand on your machine?

JunHill, 2022-02-01:
yes and yes.

I installed cmake with:
> pip3 install cmake

after it says that i don't have one

jagdeepsb, 2022-02-01:
I think that might not be sufficient. I'd recommend installing cmake [here](https://cmake.org/download/) and then adding `C:\Program Files\CMake\bin` (or the relevant directory) to your path variable.

jagdeepsb, 2022-02-01:
Sorry I take that back, it seems like you can run cmake from the terminal. Another potential issue: are you using the 2017 version of Visual Studio Build Tools`https://visualstudio.microsoft.com/vs/older-downloads/`? The 2019 version in particular is currently not supported.

JunHill, 2022-02-01:
Thank you for your quick reply! I was using the version 2019 of VS build tools. 
After changing VS build tools to 2017 version and deleting the caches in the build folder, I can now install the simulator and run gym_test.py.

closed 2022-02-01 by JunHill