EvolutionGym/evogym #3: Could NOT find GLEW (missing: GLEW_INCLUDE_DIRS GLEW_LIBRARIES)

applibres, 2022-02-15:
Hi, 
I have tried to install evogym in ubuntu 20.04. I also installed Cmaker (3.16.3). However, I got the next error message when I run "python setup.py install":

running install
running bdist_egg
running egg_info
writing evogym.egg-info/PKG-INFO
writing dependency_links to evogym.egg-info/dependency_links.txt
writing top-level names to evogym.egg-info/top_level.txt
reading manifest file 'evogym.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'evogym.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:21 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find GLEW (missing: GLEW_INCLUDE_DIRS GLEW_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindGLEW.cmake:207 (find_package_handle_standard_args)
  CMakeLists.txt:22 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Develop/evogym-main/build/temp.linux-x86_64-3.7/CMakeFiles/CMakeOutput.log".
Traceback (most recent call last):
  File "setup.py", line 72, in <module>
    zip_safe=False,
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 32, in run
    self.build_extension(ext)
  File "setup.py", line 57, in build_extension
    subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env)
  File "/Develop/anaconda3/envs/evogym/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '/Develop/evogym-main/evogym/simulator', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/Develop/evogym-main/build/lib.linux-x86_64-3.7/evogym', '-DPYTHON_EXECUTABLE=/Develop/anaconda3/envs/evogym/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.

Please, could you suggest how to solve it?
Thanks

yunshengtian, 2022-02-16:
Have you also cloned all the submodules of this repo using "git clone --recurse-submodules"?

federico-camerota, 2022-02-16:
I got the same error, `sudo apt-get install libglew-dev` solved the problem for me on Ubuntu 20.04

applibres, 2022-02-16:
Hi Everyone!
Thanks, it is working!

closed 2022-02-16 by applibres