EvolutionGym/evogym #5: Failed to find module evogym.simulator_cpp

Titanium2099, 2022-03-19:
Hello,
When I try to run gym_test.py I get the following error:

``
PS C:\Users\Yoyo\evogym> & C:/Users/Yoyo/AppData/Local/Programs/Python/Python39/python.exe c:/Users/Yoyo/evogym/examples/gym_test.py
Traceback (most recent call last):
  File "c:\Users\Yoyo\evogym\examples\gym_test.py", line 2, in <module>
    import evogym.envs
  File "C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\lib\site-packages\evogym-1.0.0-py3.9-win-amd64.egg\evogym\__init__.py", line 2, in <module>
    from evogym.sim import EvoSim
  File "C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\lib\site-packages\evogym-1.0.0-py3.9-win-amd64.egg\evogym\sim.py", line 11, in <module>
    from evogym.simulator_cpp import Sim
ModuleNotFoundError: No module named 'evogym.simulator_cpp'
``

Can you please tell me what to do to fix this error?

Yuxing-Wang-THU, 2022-06-30:
> Hello, When I try to run gym_test.py I get the following error:
>
> `PS C:\Users\Yoyo\evogym> & C:/Users/Yoyo/AppData/Local/Programs/Python/Python39/python.exe c:/Users/Yoyo/evogym/examples/gym_test.py [...]

You need to run "python setup.py install" at first

Titanium2099, 2022-06-30:
> You need to run "python setup.py install" at first

I did and a get the same thing:

`
C:\Users\Yoyo>cd evogym

C:\Users\Yoyo\evogym>py install setup.py
C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\python.exe: can't open file
 'C:\Users\Yoyo\evogym\install': [Errno 2] No such file or directory

C:\Users\Yoyo\evogym>py setup.py install
running install
running bdist_egg
running egg_info
writing evogym.egg-info\PKG-INFO
writing dependency_links to evogym.egg-info\dependency_links.txt
writing top-level names to evogym.egg-info\top_level.txt
reading manifest file 'evogym.egg-info\SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'evogym.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
creating build\bdist.win-amd64\egg
creating build\bdist.win-amd64\egg\evogym
creating build\bdist.win-amd64\egg\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\balance.py -> build\bdist.win-amd64\
egg\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\base.py -> build\bdist.win-amd64\egg
\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\change_shape.py -> build\bdist.win-a
md64\egg\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\climb.py -> build\bdist.win-amd64\eg
g\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\flip.py -> build\bdist.win-amd64\egg
\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\jump.py -> build\bdist.win-amd64\egg
\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\manipulate.py -> build\bdist.win-amd
64\egg\evogym\envs
copying build\lib.win-amd64-3.9\evogym\envs\multi_goal.py -> build\bdist.win-amd
64\egg\evogym\envs
creating build\bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Balancer-v0.json -> build\
bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Balancer-v1.json -> build\
bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BeamSlider-v0.json -> buil
d\bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BeamToppler-v0.json -> bui
ld\bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BidirectionalWalker-v0.jso
n -> build\bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BridgeWalker-v0.json -> bu
ild\bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Carrier-v0.json -> build\b
dist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Carrier-v1.json -> build\b
dist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\CaveCrawler-v0.json -> bui
ld\bdist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Climber-v0.json -> build\b
dist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Climber-v1.json -> build\b
dist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Climber-v2.json -> build\b
dist.win-amd64\egg\evogym\envs\sim_files
copying build\lib.win-amd64-3.9\evogym\envs\sim_files\DownStepper-v0.json -> bui
[... 101 lines cut ...]
creating c:\users\yoyo\appdata\local\programs\python\python39\lib\site-packages\
evogym-1.0.0-py3.9-win-amd64.egg
Extracting evogym-1.0.0-py3.9-win-amd64.egg to c:\users\yoyo\appdata\local\progr
ams\python\python39\lib\site-packages
evogym 1.0.0 is already the active version in easy-install.pth

Installed c:\users\yoyo\appdata\local\programs\python\python39\lib\site-packages
\evogym-1.0.0-py3.9-win-amd64.egg
Processing dependencies for evogym==1.0.0
Finished processing dependencies for evogym==1.0.0

C:\Users\Yoyo\evogym>cd examples

C:\Users\Yoyo\evogym\examples>py gym_test.py
Traceback (most recent call last):
  File "C:\Users\Yoyo\evogym\examples\gym_test.py", line 2, in <module>
    import evogym.envs
  File "C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\lib\site-packages\e
vogym-1.0.0-py3.9-win-amd64.egg\evogym\__init__.py", line 2, in <module>
    from evogym.sim import EvoSim
  File "C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\lib\site-packages\e
vogym-1.0.0-py3.9-win-amd64.egg\evogym\sim.py", line 11, in <module>
    from evogym.simulator_cpp import Sim
ModuleNotFoundError: No module named 'evogym.simulator_cpp'
`

Yuxing-Wang-THU, 2022-07-04:
> > > Hello, When I try to run gym_test.py I get the following error:
> > > `PS C:\Users\Yoyo\evogym> & C:/Users/Yoyo/AppData/Local/Programs/Python/Python39/python.exe c:/Users/Yoyo/evogym/examples/gym_test.py Traceback (most recent call last): File "c:\Users\Yoyo\evogym\examples\gym_test.py", line 2, in <module> import evogym.envs File "C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\lib\site-packages\evogym-1.0.0-py3.9-win-amd64.egg\evogym\__init__.py", line 2, in <module> from evogym.sim import EvoSim File "C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\lib\site-packages\evogym-1.0.0-py3.9-win-amd64.egg\evogym\sim.py", line 11, in <module> from evogym.simulator_cpp import Sim ModuleNotFoundError: No module named 'evogym.simulator_cpp'`
> > > Can you please tell me what to do to fix this error?
> > 
> > 
> > You need to run "python setup.py install" at first
> 
> I did and a get the same thing:
> 
> ` C:\Users\Yoyo>cd evogym
> 
> C:\Users\Yoyo\evogym>py install setup.py C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\python.exe: can't open file 'C:\Users\Yoyo\evogym\install': [Errno 2] No such file or directory
> 
> C:\Users\Yoyo\evogym>py setup.py install running install running bdist_egg running egg_info writing evogym.egg-info\PKG-INFO writing dependency_links to evogym.egg-info\dependency_links.txt writing top-level names to evogym.egg-info\top_level.txt reading manifest file 'evogym.egg-info\SOURCES.txt' adding license file 'LICENSE' writing manifest file 'evogym.egg-info\SOURCES.txt' installing library code to build\bdist.win-amd64\egg running install_lib running build_py running build_ext creating build\bdist.win-amd64\egg creating build\bdist.win-amd64\egg\evogym creating build\bdist.win-amd64\egg\evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\balance.py -> build\bdist.win-amd64 egg\evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\base.py -> build\bdist.win-amd64\egg \evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\change_shape.py -> build\bdist.win-a md64\egg\evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\climb.py -> build\bdist.win-amd64\eg g\evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\flip.py -> build\bdist.win-amd64\egg \evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\jump.py -> build\bdist.win-amd64\egg \evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\manipulate.py -> build\bdist.win-amd 64\egg\evogym\envs copying build\lib.win-amd64-3.9\evogym\envs\multi_goal.py -> build\bdist.win-amd 64\egg\evogym\envs creating build\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Balancer-v0.json -> build bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Balancer-v1.json -> build bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BeamSlider-v0.json -> buil d\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BeamToppler-v0.json -> bui ld\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BidirectionalWalker-v0.jso n -> build\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\BridgeWalker-v0.json -> bu ild\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Carrier-v0.json -> build\b dist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Carrier-v1.json -> build\b dist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\CaveCrawler-v0.json -> bui ld\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Climber-v0.json -> build\b dist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Climber-v1.json -> build\b dist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Climber-v2.json -> build\b dist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\DownStepper-v0.json -> bui ld\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Flipper-v0.json -> build\b dist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\GapJumper-v0.json -> build \bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Hurdler-v0.json -> build\b dist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Jumper-v0.json -> build\bd ist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\Lifter-v0.json -> build\bd ist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\ObstacleTraverser-v0.json -> build\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\ObstacleTraverser-v1.json -> build\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\package.json -> build\bdis t.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\peg.json -> build\bdist.wi n-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_files\PlatformJumper-v0.json -> build\bdist.win-amd64\egg\evogym\envs\sim_files copying build\lib.win-amd64-3.9\evogym\envs\sim_fil [... cut ...] s\Python\Python39\lib\site-packages\e vogym-1.0.0-py3.9-win-amd64.egg\evogym__init__.py", line 2, in from evogym.sim import EvoSim File "C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\lib\site-packages\e vogym-1.0.0-py3.9-win-amd64.egg\evogym\sim.py", line 11, in from evogym.simulator_cpp import Sim ModuleNotFoundError: No module named 'evogym.simulator_cpp' `

Did you download the source code by git command or zip file? If zip , some files(/evogym/simulator/external) may not be downloaded

Titanium2099, 2022-07-04:
git command, shou

> I did and a get the same thing:
>
> ` C:\Users\Yoyo>cd evogym
>
> C:\Users\Yoyo\evogym>py install setup.py C:\Users\Yoyo\AppData\Local\Programs\Python\Python39\python.exe: can't open file [...]

git, and (/evogym/simulator/external does exist , should I do it via zip?

ndr09, 2022-07-21:
Hi, I don't know if  you still need help, but I had the same problem and I found a solution (which works on windows).
In case you find a possible solution [here](https://github.com/EvolutionGym/evogym/issues/13)

jagdeepsb, 2024-08-10:
I've recently updated, simplified, and modernized `EvoGym`'s build, which should fix this problem. Feel free to reopen this issue if the problem persists!

closed 2024-08-10 by jagdeepsb