EvolutionGym/evogym #12: Error in running python gym_test.py

shuleiLiu, 2022-07-05:
In a terminal, when i cd to the examples folder and run gym_test.py, I get

Using Evolution Gym Simulator v2.2.5
Error initializing GLFW.
Segmentation fault (core dumped)

My OS is Ubuntu 18.04.4 LTS.

Yuxing-Wang-THU, 2022-07-07:
> In a terminal, when i cd to the examples folder and run gym_test.py, I get
> 
> Using Evolution Gym Simulator v2.2.5 Error initializing GLFW. Segmentation fault (core dumped)
> 
> My OS is Ubuntu 18.04.4 LTS.

I think you can delete the code "env.render()" to disable rendering the environment.

mertan-a, 2022-07-10:
I had the same error when I tried to run the example on a headless server. The solution I came up with is to use xvfb. It can be installed with `sudo apt-get install xvfb`. Then you can run the script with `xvfb-run python gym_test.py`. I believe xvfb creates a virtual x server and rendering is done using that.

jagdeepsb, 2023-02-12:
@Yuxing-Wang-THU is exactly right. EvoGym was designed so that the rendering libraries are initialized on the first call to `.render()`. This was done so that training can be run on a headless server.

Thanks @mertan-a for the cool solution!

Closing this for now, please let me know if further issues persist.

closed 2023-02-12 by jagdeepsb

JohnStewardson, 2024-03-06:
I have a similar issue, however,` xvfb-run python gym_test.py` didn't solve it. (I am using WSL 2,  Ubuntu 22.04.4 LTS)
This is the error (with and without xvfb):

Using Evolution Gym Simulator v2.2.5
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
GLFW failed to create rendering winidow.
Segmentation fault

But `apt-file search swrast_dri.so` shows:

libgl1-mesa-dri: /usr/lib/x86_64-linux-gnu/dri/kms_swrast_dri.so
libgl1-mesa-dri: /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so

Thank you for your help