EvolutionGym/evogym #13: ModuleNotFoundError: No module named 'evogym.simulator_cpp'`

ndr09, 2022-07-06:
I followed the instructions and installed Evogym on my pc (win 11). 
However, I have the following two problems  and for both I will use the _evogym/tutorials/basic_api.py_ file as reference:
1) from pycharm it give the following error, while from the terminal it executes correctly
   `Traceback (most recent call last):
  File "C:/Users/opuse/Desktop/evogym/tutorials/basic_api.py", line 1, in <module>
    from evogym.envs import *
  File "C:\Users\opuse\Desktop\evogym\evogym\__init__.py", line 2, in <module>
    from evogym.sim import EvoSim
  File "C:\Users\opuse\Desktop\evogym\evogym\sim.py", line 9, in <module>
    import evogym.simulator_cpp
ModuleNotFoundError: No module named 'evogym.simulator_cpp'`
2) If I move the basic_api.py out from the tutorials folder, hence in _evogym/basic_apy.py_, it will not execute returning the same error from 1

Thanks,
Andrea

Titanium2099, 2022-07-06:
Same as #5 , everyone with windows hits this error

Yuxing-Wang-THU, 2022-07-07:
> Same as #5 , everyone with windows hits this error

I have successfully installed Evogym on Ubuntu and Centos, but I don't know how to fix this issuse on Windows.

d9w, 2023-04-20:
This can happen if python looks for `evogym` in the current directory (for example, the source code), and not in its installed location in `dist-packages`. Notably, this will happen when using `python setup.py install` instead of `pip install .`. The compiled file `simulator_cpp.py` is only made in the `dist-packages` version

You can copy the generated `simulator_cpp.py` from the `dist-packages` installed version into your local directory or copy the full version locally if you used `python setup.py install`. However, using `pip` is probably safer and is possible with the modifications proposed in #27.

jagdeepsb, 2024-08-10:
I've recently updated, simplified, and modernized `EvoGym`'s build, which should fix this problem. Feel free to reopen this issue if the problem persists!

closed 2024-08-10 by jagdeepsb