EvolutionGym/evogym #15: compile error

TypeFloat, 2022-07-31:
I am using macOS on m1 chip, when I run the command "python setup.py install", it occurred  error:" calling a private constructor of class 'Eigen::Ref<Eigen::Matrix<double, -1, -1, 0>, 0, Eigen::OuterStride<-1>>'
                return empty;"

what should I do to fix the bug?

Hs-Cheng, 2022-11-07:
I also got the same error. The attached picture shows the error message.

[image]

Hs-Cheng, 2022-11-07:
@TypeFloat 
Do you know how to solve the problem? Thank you.

TypeFloat, 2022-11-07:
> @TypeFloat
> Do you know how to solve the problem? Thank you.

Sorry, I have not found a solution yet.

veerendrav, 2023-01-17:
@TypeFloat @Hs-Cheng  have you found the solution?

Hs-Cheng, 2023-01-17:
@veerendrav 
I have no solution, sorry about that.

amitrab, 2023-02-09:
It happened to me too on macOS.
Here's a simple solution that works:
Change those type definitions from `Ref <MatrixXd>` to `Matrix <double, 2, Dynamic>`
In directory `evogym/simulator/SimulatorCPP/` edit files:
1. `Environment.h`
2. `Environment.cpp`
3. `Sim.h`
4. `Sim.cpp`

jagdeepsb, 2023-02-12:
Hi all, thank you for your interest in EvoGym. I'm trying to recreate this issue, and suspect it has to with the cpp compiler/version of cpp being compiled.

For instance, if you are on windows, you will get this error if you attempt to build with the MS Visual Studios 2019 build tools. You will need to install the 2017 version as advised [here](http://localhost:4000/tutorials/getting-started.html#requirements).

Could someone please send me the full output of their `python setup.py install` so I can see what compiler is being used by cmake?

nakanoyusuke-cgp, 2023-02-23:
Hello.

I had the same error on MacOS with M1 chip and was able to resolve it.
Probably the reason is that the front-end compiler used by default for CMake on MacOS is the one shipped with XCode.
I was able to avoid this error by installing gcc from homebrew and specifying it as the compiler in setup.py.

(I apologize for any unnatural parts of the text, as I used a translation service to create the text.)

nakanoyusuke-cgp, 2023-02-24:
Sorry, one correction about the above message: you need to set `g++` as the C++ compiler, but do not explicitly specify the C compiler; specifying `gcc` as the C compiler seems to cause a different error.

The specific procedure is
1. install gcc and g++ (from homebrew, for example).
2. add the following statement to line 51 of `setup.py`. (replace `{path_of_g++}` with the `g++` installation path. in my case it is `/opt/homebrew/bin/g++-12`)
```
cmake_args += ["-DCMAKE_CXX_COMPILER={path_of_g++}"]
```
3. run `python setup.py install`.

\
For reference, I am attaching the full output of the `python setup.py install` before and after the problem was solved. (The user name and computer name are replaced with different strings.)

<details>
<summary>before</summary>

```
(evogym) me@mycomputer evogym % python setup.py install
running install
/Users/me/miniforge3/envs/evogym/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/Users/me/miniforge3/envs/evogym/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
creating evogym.egg-info
writing evogym.egg-info/PKG-INFO
writing dependency_links to evogym.egg-info/dependency_links.txt
writing top-level names to evogym.egg-info/top_level.txt
writing manifest file 'evogym.egg-info/SOURCES.txt'
reading manifest file 'evogym.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'evogym.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-11.1-arm64/egg
running install_lib
running build_py
creating build
creating build/lib.macosx-11.1-arm64-cpython-38
creating build/lib.macosx-11.1-arm64-cpython-38/evogym
copying evogym/world.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym
copying evogym/sim.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym
copying evogym/viewer.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym
copying evogym/__init__.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym
copying evogym/utils.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym
creating build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/climb.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/jump.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/__init__.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/traverse.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/manipulate.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/walk.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/flip.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/multi_goal.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/change_shape.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/balance.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
copying evogym/envs/base.py -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs
creating build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/DownStepper-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/rigid_3x3.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Lifter-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Traverser-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Jumper-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Balancer-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/BidirectionalWalker-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/GapJumper-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Thrower-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Balancer-v1.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Climber-v2.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/UpStepper-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
copying evogym/envs/sim_files/Walker-v0.json -> build/lib.macosx-11.1-arm64-cpython-38/evogym/envs/sim_files
[... 703 lines cut ...]
copying evogym.egg-info/dependency_links.txt -> build/bdist.macosx-11.1-arm64/egg/EGG-INFO
copying evogym.egg-info/not-zip-safe -> build/bdist.macosx-11.1-arm64/egg/EGG-INFO
copying evogym.egg-info/top_level.txt -> build/bdist.macosx-11.1-arm64/egg/EGG-INFO
writing build/bdist.macosx-11.1-arm64/egg/EGG-INFO/native_libs.txt
creating dist
creating 'dist/evogym-1.0.0-py3.8-macosx-11.1-arm64.egg' and adding 'build/bdist.macosx-11.1-arm64/egg' to it
removing 'build/bdist.macosx-11.1-arm64/egg' (and everything under it)
Processing evogym-1.0.0-py3.8-macosx-11.1-arm64.egg
[... cut ...]
Processing dependencies for evogym==1.0.0
Searching for evogym==1.0.0
Reading https://pypi.org/simple/evogym/
Couldn't find index page for 'evogym' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
No local packages or working download links found for evogym==1.0.0
error: Could not find suitable distribution for Requirement.parse('evogym==1.0.0')
```
</details>

(I apologize for any unnatural parts of the text, as I used a translation service to create the text.)

TakesxiSximada, 2023-03-18:
Hi,  I had the same errors too.

I think about evogym install process probrems has 2 factors(at least on macOS).

---

### type error

The first one, This simulator has inconsistent type definitions maybe.
The line 271 of evogym/simulator/SimulatorCPP/Environment.cpp: There is a discrepancy in the type declaration in Eigen::Ref<Eigen::Matrix<double, -1, -1, 0>, 0, Eigen::OuterStride<-1>> This topic has already been mentioned here.

https://github.com/EvolutionGym/evogym/issues/15#issuecomment-1424633656

Actually I ran into a compilation error, fixed it as per his advice. And I was able to fix the compilation error.
https://github.com/TakesxiSximada/evogym/commit/58c26593cf511fe3d8d4606b20180a8ddc6396b0 

I'm not sure if this fix is reasonable, But at least I know that can work around the type error.

---

### CMake version

The second factor is CMake version.

The latest version of CMake is 3.25.3, but the build fails evogym with this version. FindGLFW.cmake packaged with CMake seems to be the cause of the problem.

I've tried to build with some versions of CMake.

```
| cmake version | Original source | Patched source |
|---------------+-----------------+----------------|
|        3.22.2 | x               | o              |
|        3.24.2 | x               | o              |
|        3.25.1 | x               | x              |
|        3.25.3 | x               | x              |
```

(*) "Patched source" means https://github.com/TakesxiSximada/evogym/commit/58c26593cf511fe3d8d4606b20180a8ddc6396b0 has been applied.

closed 2023-07-25 by TypeFloat