EvolutionGym/evogym #16: Pip/Conda install error

aespielberg, 2022-08-07:
Having the following issues trying to install one of the submodules :).  Is it private/missing?
```
$ pip install -r requirements.txt
Collecting GPyOpt@ git+git://github.com/yunshengtian/GPyOpt@5fc1188ffdefea9a3bc7964a9414d4922603e904
  Cloning git://github.com/yunshengtian/GPyOpt (to revision 5fc1188ffdefea9a3bc7964a9414d4922603e904) to /tmp/pip-install-4qlsnhhh/gpyopt_ad9a723ba6aa450f823705c6b10c1588
  Running command git clone --filter=blob:none --quiet git://github.com/yunshengtian/GPyOpt /tmp/pip-install-4qlsnhhh/gpyopt_ad9a723ba6aa450f823705c6b10c1588
  fatal: unable to connect to github.com:
  github.com[0: 140.82.113.3]: errno=Connection timed out

  error: subprocess-exited-with-error
  
  × git clone --filter=blob:none --quiet git://github.com/yunshengtian/GPyOpt /tmp/pip-install-4qlsnhhh/gpyopt_ad9a723ba6aa450f823705c6b10c1588 did not run successfully.
  │ exit code: 128
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× git clone --filter=blob:none --quiet git://github.com/yunshengtian/GPyOpt /tmp/pip-install-4qlsnhhh/gpyopt_ad9a723ba6aa450f823705c6b10c1588 did not run successfully.
│ exit code: 128
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

LiuShuai26, 2022-08-12:
try this: 
`git config --global url."https://github.com".insteadOf git://github.com`

veerendrav, 2023-01-17:
This worked for me. Just replace git+git in requirements.txt with git+https.

jagdeepsb, 2023-02-12:
Hi @aespielberg, thank you for your interest in EvoGym.

I was able to recreate the issue, and as suggested by @LiuShuai26, and @veerendrav, was able to fix it by running

`pip install git+https://github.com/yunshengtian/GPyOpt@5fc1188ffdefea9a3bc7964a9414d4922603e904 --no-cache-dir`

Please let me know if this solves your problem and let me know if you have any further questions!

closed 2023-02-12 by jagdeepsb