EvolutionGym/evogym #19: about observation and action vectors

yyang8891, 2022-09-16:
Hi all,
    I find the env simply outputs a vec of state observation and accepts action vec of actuators (mostly a subset of all voxels), while both of them (corner points and all voxels) are arrange in matrix format by default. So may I ask if anyone could please make it clear how are the two matrices converted into vectors? How do the 2-D coordinates correspond to vector indexes? Are they simply aligned from the top-left corner to down-right element like most of the cases in image processing treatments?
    I'll be very gratefull if anyone give me a hint, thanks in advance!

Yuxing-Wang-THU, 2022-11-24:
Hi, you can refer to this page for more details about the observarion (https://github.com/Yuxing-Wang-THU/ModularEvoGym)

jagdeepsb, 2024-08-10:
Hi @yyang8891 `EvoGym` does not make any guarantees about the order of corner points or order of actuators in the vectors except that the order doesn't change across environments and across time. The implementation I've used likely returns both in top-down, left-to-right order excluding repeats.

closed 2024-08-10 by jagdeepsb