EvolutionGym/evogym #22: missing tutorials

nedelec, 2022-10-28:
The linked page 
https://evolutiongym.github.io/tutorials
Is empty: there is no tutorial.

I was unable to install on MacOS Monterey. 

> python3 setup.py install
running install
error: can't create or remove files in install directory

>  pip3 install -r requirements.txt
Collecting GPy==1.10.0
  Using cached GPy-1.10.0-cp39-cp39-macosx_10_9_x86_64.whl (1.5 MB)
ERROR: Could not find a version that satisfies the requirement gpyopt (unavailable) (from versions: 0.1.1, 0.1.2, 0.1.3, 0.1.4, 1.0.2, 1.0.3, 1.2.0, 1.2.1, 1.2.5, 1.2.6)
ERROR: No matching distribution found for gpyopt (unavailable)

Yuxing-Wang-THU, 2022-11-03:
Hi, I think you can install the GpyOpt manully by downloading files from this link 

https://github.com/yunshengtian/GPyOpt/tree/5fc1188ffdefea9a3bc7964a9414d4922603e904 

then use python setup.py  install

I used to solve this problem by doing so.

jagdeepsb, 2023-02-12:
Hi @nedelec, thank you for your interest in EvoGym. 

I believe the issue with `GPyOpt` is that there is already a cached version on your device which is being used by `pip`. Try running:

`pip install git+https://github.com/yunshengtian/GPyOpt@5fc1188ffdefea9a3bc7964a9414d4922603e904 --no-cache-dir`

Or

`pip install git+git://github.com/yunshengtian/GPyOpt@5fc1188ffdefea9a3bc7964a9414d4922603e904 --no-cache-dir`

Regarding tutorials, you can access them in the navbar on the left of the [site](https://evolutiongym.github.io/tutorials). I’ve also recently updated the website to make it easier for new users to navigate through, and fixed the broken tables of contents on a couple of pages.

Please let me know if you have any further questions!

closed 2023-03-23 by jagdeepsb