EvolutionGym/evogym #23: I also got the same error. The attached picture shows the error message.

veerendrav, 2023-01-17:
I also got the same error. The attached picture shows the error message.

[image]

_Originally posted by @Hs-Cheng in https://github.com/EvolutionGym/evogym/issues/15#issuecomment-1304940827_

jagdeepsb, 2023-02-12:
Hi @veerendrav thanks for your interest in EvoGym. I'm trying to recreate this issue, and suspect it has to with the cpp compiler/version of cpp being compiled.

For instance, if you are on windows, you will get this error if you attempt to build with the MS Visual Studios 2019 build tools. You will need to install the 2017 version as advised [here](http://localhost:4000/tutorials/getting-started.html#requirements).

Could you please send me the details of your system as well as the full output of your `python setup.py install`

jrieffel, 2023-07-17:
Hi there I'm getting this same error as well.

OS: MacOS 13.13.1
CPU: arm64 (M1 Pro)
CMake: cmake version 3.25.3
Compiler: Apple clang version 14.0.3 (clang-1403.0.22.14.1)

Full output attached as textfile.
[output.txt](https://github.com/EvolutionGym/evogym/files/12068332/output.txt)

jagdeepsb, 2024-08-10:
I've recently updated, simplified, and modernized `EvoGym`'s build, which should fix this problem. Feel free to reopen this issue if the problem persists!

closed 2024-08-10 by jagdeepsb