EvolutionGym/evogym #26: Gym dependency

d9w, 2023-04-20:
Evogym currently depends on gym 0.19.0, however pip no longer supports installing this version 

```
pip install gym==0.19.0
Collecting gym==0.19.0
  Using cached gym-0.19.0.tar.gz (1.6 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in gym setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.
```

More recent versions of gym (>0.26.0) have changed their API and in the future, gym will be replaced by [gymnasium](https://github.com/Farama-Foundation/Gymnasium), which uses the gym>0.26.0 API. Would it be possible to update the dependency to gymnasium and change the API?

For now, I've been able to use evogym with version 0.22.0 of gym, however this breaks the requirement to have stable-baselines3==1.4.0 in `requirements.txt`. This requirement doesn't seem necessary to run evogym, just to run the examples.

Okabe-Junya, 2024-07-18:
Hello @d9w!

FYI: #35 would be useful for you, thank you!

jagdeepsb, 2024-08-10:
Updated in [v2.0.0](https://github.com/EvolutionGym/evogym/releases/tag/2.0.0). Feel free to reopen this issue if the problem persists!

closed 2024-08-10 by jagdeepsb