EvolutionGym/evogym #33: Windows installation unable to run GA.

PFLigthart, 2023-11-27:
I have attempted to install in Windows using conda as described in the README.

When attempting to install I followed the step-by-step in the README religiously. I am on Windows 10 Home version 22H2.

When running `conda env create -f environment.yml` with conda version 23.7.2 the environment seems to be created successfully. Then after activating the environment and running `python setup.py install` it completes successfully with some warnings (see snippet in error output B below). Running `python gym_test.py` works as expected however running `python .\run_ga.py` results in (see error output A below). Attempting to rectify this was unsuccessful.

Error output A:
----------------
Traceback (most recent call last):
  File ".\run_ga.py", line 4, in <module>
    from ga.run import run_ga
  File "C:\mydocs\github\evo_gym_errors\evogym\examples\ga\run.py", line 14, in <module>
    from ppo import run_ppo
  File "C:\mydocs\github\evo_gym_errors\evogym\examples\ga\..\ppo\__init__.py", line 2, in <module>
    from ppo.run import run_ppo
  File "C:\mydocs\github\evo_gym_errors\evogym\examples\ga\..\ppo\run.py", line 9, in <module>
    from ppo import utils
  File "C:\mydocs\github\evo_gym_errors\evogym\examples\ga\..\ppo\utils.py", line 7, in <module>
    from ppo.envs import VecNormalize
  File "C:\mydocs\github\evo_gym_errors\evogym\examples\ga\..\ppo\envs.py", line 7, in <module>
    from stable_baselines3.common.monitor import Monitor
ModuleNotFoundError: No module named 'stable_baselines3'

Error output B:
----------------
running install
C:\Miniconda\envs\evogym\lib\site-packages\setuptools\command\install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
C:\Miniconda\envs\evogym\lib\site-packages\setuptools\command\easy_install.py:147: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  EasyInstallDeprecationWarning,
. . . 
  SimObject.cpp
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(49): warning C4244: 'argument': convers
ion from 'double' to 'GLsizei', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyt
hon-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(52): warning C4244: 'argument': convers
ion from 'double' to 'size_t', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyth
on-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(91): warning C4244: 'argument': convers
ion from 'double' to 'GLsizei', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyt
hon-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(96): warning C4244: 'argument': convers
ion from 'double' to 'size_t', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyth
on-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(121): warning C4244: 'argument': conver
sion from 'double' to 'GLsizei', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpy
thon-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(125): warning C4244: 'argument': conver
sion from 'double' to 'size_t', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyt
hon-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(133): warning C4244: 'return': conversi
on from 'double' to 'int', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpython-3
7\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(136): warning C4244: 'return': conversi
on from 'double' to 'int', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpython-3
7\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Camera.cpp(144): warning C4244: 'return': conversi
on from 'double' to 'int', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpython-3
7\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Boxel.cpp(24): warning C4244: 'argument': conversi
on from 'int' to 'const float', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyt
hon-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\Boxel.cpp(25): warning C4244: 'argument': conversi
on from 'int' to 'const float', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyt
hon-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\SimObject.cpp(30): warning C4244: 'argument': conv
ersion from 'double' to 'int', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-cpyth
on-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\SimObject.cpp(31): warning C4244: 'argument': conv
ersion from 'double' to 'const unsigned __int64', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\t
[... 390 lines cut ...]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\PhysicsEngine.cpp(945): warning C4244: 'argument':
 conversion from 'float' to 'const unsigned __int64', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\bui
ld\temp.win-amd64-cpython-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\PhysicsEngine.cpp(950): warning C4244: 'argument':
 conversion from 'float' to 'int', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-c
python-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\PhysicsEngine.cpp(957): warning C4244: 'argument':
 conversion from 'float' to 'const _Ty', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-a
md64-cpython-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
          with
          [
              _Ty=int
          ]
C:\mydocs\github\evo_gym_errors\evogym\evogym\simulator\SimulatorCPP\PhysicsEngine.cpp(958): warning C4244: 'argument':
 conversion from 'float' to 'int', possible loss of data [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd64-c
python-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
LINK : warning LNK4044: unrecognized option '/ldl'; ignored [C:\mydocs\github\evo_gym_errors\evogym\build\temp.win-amd6
4-cpython-37\Release\SimulatorCPP\simulator_cpp.vcxproj]
     Creating library C:/mydocs/github/evo_gym_errors/evogym/build/temp.win-amd64-cpython-37/Release/SimulatorCPP/Relea
  se/simulator_cpp.lib and object C:/mydocs/github/evo_gym_errors/evogym/build/temp.win-amd64-cpython-37/Release/Simula
  torCPP/Release/simulator_cpp.exp
  Generating code
  Finished generating code

. . .

lucyivt, 2024-06-01:
I typed `pip install stable-baselines3[extra]` into the terminal from the examples directory. This solved the first error 
"from stable_baselines3.common.monitor import Monitor
ModuleNotFoundError: No module named 'stable_baselines3"
However now I'm getting a new error `gym.error.NameNotFound: Environment `roboticgamedesign` doesn't exist.`

jagdeepsb, 2024-08-10:
@PFLigthart @lucyivt 
I've recently updated, simplified, and modernized `EvoGym`'s build, which should fix this problem. Feel free to reopen this issue if the problem persists!

closed 2024-08-10 by jagdeepsb