EvolutionGym/evogym #34: NumPy attribute Error

csvelayutham, 2024-03-16:
Hi,

I installed EvoGym in Linux Mint having Python 3.10. As suggested in the github page I tried to execute the gym_test.py in examples. It issues the following error.

Using Evolution Gym Simulator v2.2.5
Traceback (most recent call last):
File "/home/cs_velayutham/Documents/Koding/EvoRo/evogym/examples/gym_test.py", line 9, in
env = gym.make('Walker-v0', body=body)
File "/home/cs_velayutham/.local/lib/python3.10/site-packages/gym/envs/registration.py", line 640, in make
env = env_creator(**_kwargs)
File "/usr/local/lib/python3.10/dist-packages/evogym-1.0.0-py3.10-linux-x86_64.egg/evogym/envs/walk.py", line 29, in init
self.action_space = spaces.Box(low= 0.6, high=1.6, shape=(num_actuators,), dtype=np.float)
File "/home/cs_velayutham/.local/lib/python3.10/site-packages/numpy/init.py", line 324, in getattr
raise AttributeError(former_attrs[attr])
AttributeError: module 'numpy' has no attribute 'float'.
np.float was a deprecated alias for the builtin float. To avoid this error in existing code, use float by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use np.float64 here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations. Did you mean: 'cfloat'?

Request for help please

csvelayutham, 2024-03-16:
Tried downgrading the numpy version as follows

!pip install --upgrade numpy===1.22.0

When I tried python3 gym_test.py I got the following errors

------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Using Evolution Gym Simulator v2.2.5
/home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:174: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed a `seed` instead of using `Env.seed` for resetting the environment random number generator.
  logger.warn(
/home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:187: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed `options` to allow the environment initialisation to be passed additional information.
  logger.warn(
/home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:195: UserWarning: WARN: The result returned by `env.reset()` was not a tuple of the form `(obs, info)`, where `obs` is a observation and `info` is a dictionary containing additional information. Actual type: `<class 'numpy.ndarray'>`
  logger.warn(
/home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:219: DeprecationWarning: WARN: Core environment is written in old step API which returns one bool instead of two. It is recommended to rewrite the environment with new step API. 
  logger.deprecation(
Traceback (most recent call last):
  File "/home/cs_velayutham/Documents/Koding/EvoRo/evogym/examples/gym_test.py", line 14, in <module>
    ob, reward, done, info = env.step(action)
  File "/home/cs_velayutham/.local/lib/python3.10/site-packages/gym/wrappers/time_limit.py", line 50, in step
    observation, reward, terminated, truncated, info = self.env.step(action)
ValueError: not enough values to unpack (expected 5, got 4)

Request for help please.

ShuleiLiu, 2024-03-19:
I also got the same error. Request for help please.

Yuxing-Wang-THU, 2024-06-04:
> Tried downgrading the numpy version as follows
> 
> !pip install --upgrade numpy===1.22.0
> 
> When I tried python3 gym_test.py I got the following errors
> 
> Using Evolution Gym Simulator v2.2.5 /home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:174: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed a `seed` instead of using `Env.seed` for resetting the environment random number generator. logger.warn( /home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:187: UserWarning: WARN: Future gym versions will require that `Env.reset` can be passed `options` to allow the environment initialisation to be passed additional information. logger.warn( /home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:195: UserWarning: WARN: The result returned by `env.reset()` was not a tuple of the form `(obs, info)`, where `obs` is a observation and `info` is a dictionary containing additional information. Actual type: `<class 'numpy.ndarray'>` logger.warn( /home/cs_velayutham/.local/lib/python3.10/site-packages/gym/utils/passive_env_checker.py:219: DeprecationWarning: WARN: Core environment is written in old step API which returns one bool instead of two. It is recommended to rewrite the environment with new step API. logger.deprecation( Traceback (most recent call last): File "/home/cs_velayutham/Documents/Koding/EvoRo/evogym/examples/gym_test.py", line 14, in ob, reward, done, info = env.step(action) File "/home/cs_velayutham/.local/lib/python3.10/site-packages/gym/wrappers/time_limit.py", line 50, in step observation, reward, terminated, truncated, info = self.env.step(action) ValueError: not enough values to unpack (expected 5, got 4)
> 
> Request for help please.

The following configs work for me, I also got the same error when using python 3.10.

python 3.7.11 
numpy 1.21.5 
 gym  0.19.0

I think the current evogym dependencies need to be updated to support higher python version.

jagdeepsb, 2024-08-10:
@csvelayutham @ShuleiLiu @Yuxing-Wang-THU 
I've recently updated, simplified, and modernized `EvoGym`'s dependencies, which should fix this problem -- (numpy `<2.0.0` now supported). Feel free to reopen this issue if the problem persists!

closed 2024-08-10 by jagdeepsb