EvolutionGym/evogym #48: Maximum size of robot 14x14

LeoneCosti96, 2024-10-23:
When playing around with the simulation, I found out that if you make the robot NxN, everything works perfectly up to N=14, but the simulation does not even start from N=15. Is it due to computation time, or is there some intrinsic limit in the way the simulator is implemented? I am rendering in 'screen'mode

jagdeepsb, 2024-10-24:
Hi! There should not be any such limitation. Is it just silently failing? Have you tried different robot designs?

LeoneCosti96, 2024-10-24:
Thank you for your reply! I have performed a better investigation using sample_robot() multiple times, and at an increasing N. it does work but as N increases the time before the beginning of the rendering increases significantly. It becomes a problem if trying to optimize big robots, but I am not sure what can be done about it.

jagdeepsb, 2024-10-24:
Is the initial load time slow, or is the sim.step() slow (or both)? how long is each taking?

jagdeepsb, 2025-06-07:
Hey, I wasn't able to recreate this issue. Could you send me a robot structure that is taking a while for you to load?