EvolutionGym/evogym #49: Changing simulation parameters such as gravity and the inter-step time interval

LeoneCosti96, 2024-10-24:
Hi! It's me again. I was wondering if it is possible to add forces acting on the robot, or in general changing physical variables in the environment. I have found the definition of gravity = 110 in evogym/simulator/SimulatorCPP/PhysicsEngine.h, but ideally I would like to be able to control that. Also, why is it 110? What is the unit of measure? Another thing I would love to be able to do is to change the simulation time between each "step", is is something that can be done through the API? Thank you so much for your help

jagdeepsb, 2024-10-24:
Hey EvoGym does not have an API that supports changing physics params. The values here have no meaning/units -- they have been empirically tuned to achieve reasonable-looking simulation for common robot sizes. 

If necessary, you can experiment with different physics params by changing the underlying simulator code and rebuilding it.

leonbaiyu, 2025-01-20:
Hey jagdeepsb, 

Do you have any idea of the relation between the gravity, step time and mass in this simulator, and reality. For this to be used in publishable research, it would greatly benefit from some connection to reality in the physics parameters.

jagdeepsb, 2025-02-15:
EvoGym models each robot/object as a spring-mass system.
Mass = `1.0`
Gravity = `110`

Each step of the environment calls `30` steps of the simulator (which uses RK4 integration with a timestep of `0.0001`). 

Again, these values have been empirically tuned to achieve reasonable-looking simulation for common robot sizes.

leonbaiyu, 2025-02-19:
Thanks for getting back to me jagdeepsb. And how is the effect of Gravity calculated?

jagdeepsb, 2025-02-26:
Using standard equations of physics.
I might recommend looking [here](https://github.com/EvolutionGym/evogym/blob/de71e9a30fe029ee103620d2c63745986bd6f2e1/evogym/simulator/SimulatorCPP/PhysicsEngine.cpp#L427)