EvolutionGym/evogym #50: Request to share the experiment results

pablo2811, 2024-11-17:
Hi!

I'm currently researching a new method for design optimisation and I would like to base my work on the `evogym` environment. 

Question: Could you please share the designs of the best motion-related robots and the PPO parameters used for control policy learning as well as their relative rewards? 

That way, after we develop our new approach to design optimisation we will be able to compare the results with yours. 
Thanks in advance and I hope this is a right channel for such messages.

jagdeepsb, 2025-02-15:
Hey, you can see page `38` of our [paper](https://arxiv.org/pdf/2201.09863) to see the best relative rewards achieved by our algorithm. We do not have the best designs at this time.

jagdeepsb, 2025-06-04:
Hey, if it's helpful to you, I've just released some datasets of robot designs and policies that might be helpful to your research: https://github.com/EvolutionGym/evogym-datasets

closed 2025-06-04 by jagdeepsb